Jstudner/jcorp-nomad #12: ESPAsyncWebServer.h: No such file or directory

brock, 2025-07-12:
Building in Arduino IDE is throwing this error:

```
~/jcorp-nomad/firmware/JcorpNomadProject/JcorpNomadProject.ino:4:10: fatal error: ESPAsyncWebServer.h: No such file or directory
    4 | #include "ESPAsyncWebServer.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: ESPAsyncWebServer.h: No such file or directory
```

Jstudner, 2025-07-12:
Howdy! This is an easy fix! go [here](https://github.com/Jstudner/jcorp-nomad/releases/tag/library) and download /unzip the library folder. Point Arduino at those, or clean out your library files and place these there. (on windows its usually C:\Users\<User>\Documents\Arduino\libraries

also if you don't have a libraries file yet just drop the extracted folder in Arduino and it should fix your issue!

Vandypointe2, 2025-08-16:
> Howdy! This is an easy fix! go [here](https://github.com/Jstudner/jcorp-nomad/releases/tag/library) and download /unzip the library folder. Point Arduino at those, or clean out your library files [...]

So I tried to do this, but when I drop the extracted folder in, it continually just restarts and show the projects files as jumbled. I am not really sure how to fix the libraries as I have tried to import as a zip, and it tells me that it is not a library zip correctly formatted, and I have not been able to manually place it in the sketch folder location as a libraries folder. Do you have a tutorial video I could follow in some way for the firmware flashing portion of all of this?

Jstudner, 2025-08-16:
Howdy! If you use the latest version of the code from GitHub I actually fixed the library issue all together. When did you download the actual code from GitHub? Let me know and I will help you troubleshoot!

Jstudner, 2025-08-16:
Actually I know exactly what the issue is, I just tested it my bad 😅. Still super simple you need to manually grab a few library's. Go to the library manager and you can add them there. I'll update the docs tonight with the exact versions.

Vandypointe2, 2025-08-16:
Honestly I know nothing about any of this, so its all new, but looking at the project files its apparent that its way over my head. Fingers crossed I can get it working.

Jstudner, 2025-08-16:
I should be able to get you through it without too much more trouble. Trying to make it as beginner friendly as possible so the feedback certainly helps. (Also YouTube video tutorial coming soon editing is hard 😅)

Here is what you need to get it running. On the left side panel of Arduino IDE there are 5 icons, the middle one looks like books and when clicked says "library manager" 

You need to search for and install the following library's and then it should compile fine

"ArduinoJson" by Benoit Blanchon v7.3.0
"Async TCP" by ESP32Async v3.4.7
"ESP Async Webserver" by ESP32Async v 3.7.1
"LVGL" by kisvegabor V8.4.0

Pay attention to the versions, some will work but best to keep the same ones. Install each and then try compiling again, should fix your problem!

Vandypointe2, 2025-08-17:
> I should be able to get you through it without too much more trouble. Trying to make it as beginner friendly as possible so the feedback certainly helps. (Also YouTube video tutorial coming soon [...]

I was able to get it all flashed and running. Changed the SSID and other settings. I  put an MP4 into the SD storage in the movie folder and it shows in the admin files view, just like the sample videos, but in the movies area of the main menu, there are no types of media in any of the selections, even though I know the board can see it all. Even after gen of the json. I am not sure where I am wrong here.