Jstudner/jcorp-nomad #15: Impossible to compile

alfem, 2025-07-13:
Hi!

This looks like an amazing project! I used a portable media streamer years ago, in long trips with my little sons. But this DIY solution is really interesting (and portable).

However, I can not compile it. I tried both, main an experimental branches. I moved my usual libraries away and put your zipped libraries in place.

But compile process gives these errors: 
```

/prog/jcorp-nomad/firmware/JcorpNomadProject/Display_ST7789.cpp: In function 'void Backlight_Init()':
Display_ST7789.cpp:217:3: error: 'ledcAttach' was not declared in this scope
   ledcAttach(EXAMPLE_PIN_NUM_BK_LIGHT, Frequency, Resolution);
   ^~~~~~~~~~
/prog/jcorp-nomad/firmware/JcorpNomadProject/Display_ST7789.cpp:217:3: note: suggested alternative: 'ledcAttachPin'
   ledcAttach(EXAMPLE_PIN_NUM_BK_LIGHT, Frequency, Resolution);
   ^~~~~~~~~~
   ledcAttachPin

/prog/jcorp-nomad/firmware/JcorpNomadProject/Wireless.cpp: In function 'int ble_scan_number()':
Wireless.cpp:44:51: error: cannot convert 'BLEScanResults' to 'BLEScanResults*' in initialization
   BLEScanResults *foundDevices = pBLEScan->start(5);
                                                   ^
```

The only libraries I have not replaced are the esp32/hardware ones.

Any clue?

Jstudner, 2025-07-14:
Howdy! Glad you are excited about the project! Let me see if I can help. I would start by checking you are using the same Arduino version, it shouldn't matter but I wouldn't be that surprised, I use 2.3.4. Next thing is for the board manager are you using esp32 ver 3.1.3. I havent personaly hit the same error as you before, but it seems like the newer versions use different calls, which is why you get that error. (also double check you setup the tools section correctly for the waveshare board.) let me know if that does the trick, if not ill look into it some more!

Jstudner, 2025-07-14:
any chance you are using a Tdongle board instead of the waveshare one? I just got a similar error trying to work with that board.

Jstudner, 2025-07-15:
Howdy! Check relieses on the main page, as it says in the instructibles the libraries are really specific so best to just download and use the zip included. 
https://github.com/Jstudner/jcorp-nomad/releases/tag/library

This has the exact ones to use.

Best of luck! I'll look into the lv_conf next!

weecc-git, 2025-07-15:
Thank you so much. I just copied the lv_config_template from the lvl library and renamed it lv_conf.h and dropped it into the same level as lvl, and that allowed the compile to continue. Thanks again for you work on this project!

Jstudner, 2025-07-15:
Good catch, I will update both of those! I am about to update with a new editable UI for the LCD, so had to change a bunch anyway. Thanks for letting me know!

weecc-git, 2025-07-15:
Final note: I managed to successfully compile and upload the firmware ONLY after setting the Arduino IDE parameter Flash Size to 16MB from the default 4MB. Otherwise the firmware will upload but the device will be locked in a boot loop because the partition topology is incorrect.