Jstudner/jcorp-nomad #17: [Feature Request] ZIM / Kiwix support

gjgradem, 2025-07-15:
This is an amazing project and i'm soon to build my own! What would make this even more amazing is perhaps adding support for Kiwix / zim files so it not only is an entertainment device, but also a survival device!

Cheers and good luck with this amazing project!

Jstudner, 2025-07-15:
Ill take a crack at this, Really good idea actually!

Jstudner, 2025-07-15:
Ok, yea after looking into it a bit it should be doable, but there are a few things: Fat32 limits the max individual file size to 4gb > zim files can be downloaded in a split format under 4gb each, that being said the support for file splitting looks less than ideal. Any Zim you can download under 4gb will work by default, but you may need to mess around with online tools to split larger ones, and getting full Wikipedia may not be great. Ill be working it it though! 

Also going to add a script to the sd card files for running the webserver off a PC so the data isn't stuck on the device if it gets damaged!

Jstudner, 2025-07-15:
I will put a reader for it together, its handled fully in the browser so it might run a bit slow but it will defiantly work to some extent. If I can find a reliable way to split zims I will make a guide! For now I am going to focus on basic functionality, but I think I can just yoink their frontend that already supports slits.

Jstudner, 2025-07-15:
But yes, esp32 only supports fat32, which is nice in some ways, but yea it mostly sucks lol, the bigger brother to this thing is going to run with an ARM based chip, so that will fix like... literally everything

gjgradem, 2025-07-15:
Haha you have a lot of knowledge about these ESP's. Limitations exist and that's fine. Then i think the comment you made about creating basic functionality will be more then fine. I just ordered the ESP and am very curious to see if i can install it! Amazing work man. Thanks :)

Jstudner, 2025-07-18:
That could work, but the issue is that many of my libraries don't work in exfat, its much less common for esp32. not impossible but a lot of work for a minor feature, would possibly need to scratch build some. 

As far as file splitting goes it seem like support for that has widely ended. I have seen a few tools, but many complaints about it corrupting files and not being reliable. I am still planning to get some level of ZIM support since it makes sense to have though, and there are way to split files so if I can get it reliable I will post documentation of that too!

Jstudner, 2025-07-29:
ok, I have been doing a bunch of work getting polish done on Nomad so I can update main. after revisiting this for a few hours I have decided I am not going to support kiwix just yet. I can get it to work as long as the files are under 4gb, but I cant get a reliable method for splitting larger files that doesn't break them. I will add support to the bigger version of nomad I am setting up next though! I love the idea, but it shines with a large library and I just cant get that yet.

closed 2025-07-29 by Jstudner