Jstudner/jcorp-nomad #35: Feature Request: Internet connectivity

unxplained, 2025-07-28:
Hi,

I was thinking that it would be very useful if the device can connect to a wifi network so that the client device would have internet connectivity.

E.g. hotel wifi > nomad > phone, for the phone to still have internet connectivity while connected to nomad

Thanks

mhschmall, 2025-07-29:
https://techtutorialsx.com/2021/01/04/esp32-soft-ap-and-station-modes/

Jstudner, 2025-07-29:
OK! I have done some research, and for now I don't think this is an option. The article mhshmall linked is useful, but doesn't actually let you route traffic between the two networks, so its not feasible in this case. I looked into modifying it, but it takes too much work for the CPU/RAM to manage while also doing web server stuff. I did also try looking into having the captive portal not grab and redirect every single request (which does allow for you to use mobile data), but I cant get it to work reliably. For now I am going to back burner it, I will defiantly revisit the idea when I build the big version (that should have enough power to properly send packets and use both networks as intended)