Jstudner/jcorp-nomad #39: Feature Request: Need gallery upload button

mhschmall, 2025-08-03:
This new release is amazing. You've put a lot work into it and it shows! 

However, the gallery page really needs an upload button. For nominal security, the file to be uploaded should be sanity checked against the  mediaExts list and rejected if it doesnt have a valid extension.

Thanks for all the work.

Jstudner, 2025-08-03:
Ah! They didnt upload due to being empty. I will add placeholder items to them this afternoon and update! Thx for the catch!

Jstudner, 2025-08-03:
This one will take a bit longer since I want to have some level of admin control for it (maybe disable it if the admin user doesn't want people to upload) That being said it would be cool to have. I will also need to have it check the SD card for space as well as the other sanity checks, so I will probably address it when I redo the SD card checks (causes a screen glitch on the LCD every 60 sec as is)

mhschmall, 2025-08-03:
My use case scenario was using nomad-enabled stick as a party hub: your music and play lists are already loaded, any vids you want to show are loaded, drink recipes are loaded in the books section, and gallery was where people could share snaps and vids they took during the party.

Jstudner, 2025-08-03:
That's so awesome, glad you can get some good use out of it! Make sure to get the RGB going haha 😂

Keep me posted, I really like seeing people come up with unique use cases!

Jstudner, 2025-08-03:
I don't think I am going to add this to the production version, but I'll update experimental with it once I get the main branch fixed up!

closed 2025-08-08 by Jstudner

mhschmall, 2025-08-09:
Looks good except gallery page does not refresh upon upload completion. Page sits there with the status bar at 100%.