Jstudner/jcorp-nomad #41: File browser on admin page does not refress file list after action

mhschmall, 2025-08-03:
Tested on both firefox and edge, the file list is not refreshed after most actions are performed. The only action that didnt have a problem was download, which makes sense: its pops up a dialog and then does its thing. No need to refresh the file list.  All other actions are affected.

Additionally, the progress bar squishes things left. I believe making columns fixed width solves this problem, but perhaps reconsidering the layout may be a better approach.

Now:  column, column, column 
            list           list       action/status
                                        
New? spanning cell
           action/status
        column column
           list        list

Jstudner, 2025-08-08:
Fixed in new update, cleaned up the UI, new items reload without needing to refresh the whole page. You need to re-flash the new ino, and update and admin.html with the new one.

mhschmall, 2025-08-09:
Confirmed.