Jstudner/jcorp-nomad #44: Trouble making usb mode work

wasspaul, 2025-08-04:
First off thanks for all the effort on this great project.
I am trying the latest experimental build and can not seem to make the usb mode function.

Anyone else having this issue?

I go to the admin page select USB mode.  I see the device go into usb mode (first with a dim display, then with a brighter one.)   After several seconds the device appears to reboot and goes into normal mode.

Any ideas on what I might be doing wrong would be great.

Thanks for your help.

Below is the serial monitors output.

=== Booting Nomad (debug) ===

=== ESP32-S3 Captive Portal & SDMMC Server ===
WiFi AP started...
Initializing SD Card...
SD Card initialized successfully!
Loading Settings...
[SETTINGS] autoGenerateMedia = false
Starting WiFi with SSID: PAW_Mserve
settings.brightness = 27
[BOOT] autoGenerateMedia (from settings) = false
[BOOT] generate_once.flag = false
[BOOT] Skipping media.json generation.
Web Server started!
User-Agent: Microsoft NCSI
Android device. Serving index.html
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0
Android device. Serving index.html
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0
Android device. Serving index.html
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0
Android device. Serving index.html
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0
Android device. Serving index.html
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0
Android device. Serving index.html
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0
Android device. Serving index.html
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0
Android device. Serving index.html
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0
Android device. Serving index.html
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0
Android device. Serving index.html
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0
Android device. Serving index.html
>>> /enterUsb handler hit
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037af22
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1180
load:0x403c8700,len:0xc2c
load:0x403cb700,len:0x311c
entry 0x403c88b8
=== Booting Nomad (debug) ===
>>> USB mode: mounting SD & starting MSC
>>> USB mode: mounting SD & starting MSC

Note after here I dont get serial output.  best I can tell the PC is in usb mode but the esp is not.

Im running on a windows 10 PC

wasspaul, 2025-08-04:
Thanks for the super quick response.  For me, with USB OTG it looks like its working.  I have not tested it with traffic yet but in a first look I can see it in the file explorer.   I did take a long time to show up in the windows system but yeah its working!

Thanks again.

Jstudner, 2025-08-04:
Yea, I am still working on improving the mount time, but I think to make any real improvement on USB mode I will need to switch to expressif IDE instead of Arduino. Not exactly sure why but the online people say it will make files go faster.