Jstudner/jcorp-nomad #52: Games Emulation Option / Library

Jstudner, 2025-08-13:


mhschmall, 2025-08-13:
I personally think serving rom's and emulation is a dog whistle for lawyers.

Jstudner, 2025-08-13:
> I personally think serving rom's and emulation is a dog whistle for lawyers.

Well that's a terrifying thought. But actually quite a valid point. I forget you can't buy ROMs 😅. 

For now I will stick with HTML games anyway since I don't have a good method for testing emulation anyway.

MMagTech, 2025-08-13:
I understand the legality of this from your point. If for some reason you implement it would be epic from an already awesome piece of software. Think you’d be covered though from a legality standpoint since the roms wouldn’t be supplied.