Jstudner/jcorp-nomad #64: Board Compatability - from Waveshare to goouuu esp32-s3

GitPIrate2025, 2025-08-19:
Essentially bought the wrong board for this project by accident but decided to go ahead and try to get it working. The original Gen 1 firmware required very little changes to get the server up and running. 

The Goouuu Board is a 1 bit board and its pin allocation is different. As such the following changes were made -

// SD card pinout for Goouuu esp32s3 CAM
#define SD_CLK_PIN 39
#define SD_CMD_PIN 38
#define SD_D0_PIN 40
//#define SD_D1_PIN 18
//#define SD_D2_PIN 17
//#define SD_D3_PIN 21

and 

   // Initialize SD card
    Serial.println("Initializing SD Card...");
    if (!SD_MMC.setPins(SD_CLK_PIN, SD_CMD_PIN, SD_D0_PIN)) {
        Serial.println("ERROR: SDMMC Pin configuration failed!");
        return;
    }

this ensures the board runs in 1 bit mode and listens to the correct lanes.

doing this alone allows the Goouuu board to flash and is perfectly functional. (though processing power is being used to run an LCD that is not there) 

Problem occurs when trying to use experimental firmware or the new folded firmware main. 

changes made

the changes above are replicated in the new firmware but also the RGB_lamp.h function also had to be disabled. This is due to the fact that it uses pin 38 and 40 which the Goouuu needs for SD.  Muting it and its relative functions in code allows the board to flash. 

However, this results in an apparent boot loop. I believe the watchdog is being triggered but honestly, I have no idea. This is my first dip in coding in general let alone working with ESP hardware. I'm a quick learner but not that quick.  

The error is shown here -

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376789
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1180
load:0x403c8700,len:0xc2c
load:0x403cb700,len:0x311c
entry 0x403c88b8
Set Backlight parameters in the range of 0 to 100 

Set Backlight parameters in the range of 0 to 100 
E (14428) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (14428) task_wdt:  - IDLE0 (CPU 0)
E (14428) task_wdt: Tasks currently running:
E (14428) task_wdt: CPU 0: SDScan
E (14428) task_wdt: CPU 1: IDLE1
E (14428) task_wdt: Aborting.
E (14428) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x420f0f0e:0x3fcc8730 0x4201da81:0x3fcc8750 0x42001c35:0x3fcc8770 0x42001cfd:0x3fcc87a0 0x42001d89:0x3fcc87c0 0x42010f85:0x3fcc9d70 0x420ef699:0x3fcc9d90 0x42025825:0x3fcc9dc0 0x420262dd:0x3fcc9e90 0x420389aa:0x3fcc9f10 0x4200946e:0x3fcc9f40 0x4037e789:0x3fcc9f60

this repeats infinitely. 

[JcorpNomadProject_MODIFIED.txt](https://github.com/user-attachments/files/21849791/JcorpNomadProject_MODIFIED.txt)

A copy of the code I have modified is also included.

Jstudner, 2025-08-19:
Howdy! Based on the error it looks like it's getting stuck trying to talk to the SD card. I will take a look through the code when I am home, but check the pins carefully on the actual board. 1-bit mode is also likely causing an issue, something else in my code might expect it though I will have to check, not sure off the top of my head.

GitPIrate2025, 2025-08-19:
thanks man! look I really appreciate you even bothering to look into it. This is such a cool project, and I would love to be able to use all the new features.

GitPIrate2025, 2025-09-10:
Hey mate, thanks so much for taking the time to troubleshoot. I have had a stab or two at it but could not make much headway. Good to know im in with a chance in the next update maybe. 

also all the new features your managing to cram into this project is insane. One this i was trying to do is hookup a 2.5" hdd to my board. its has OTG support. I managed to get it to recognize the drive so far but not load from it.I will provide code when its all working.

Jstudner, 2025-09-10:
I actually might be able to give you a tip on that end! It's likely not picking it up due to being exfat. The chip firmware esp32 comes with only works with fat32 drives. There is a workaround using expressifs ide, but I didn't want to complicate the tutorial as it's less friendly than Arduino. I am working on modifying the actual source files though so at some point I should be able to have exfat support. If you can convert the drive to fat32 (untested but fat32 format will probably do that) you may have better luck.

One of the things I am including on the chonky nomad (this thing is gonna take for fricking ever I keep making it more complicated) is a USB A female port that can take USB hard drives so you can have much larger / modular expandable storage. 

Excited to see if you can get this feature working, I would love to add it to the main project if it's successful! 😁

Jstudner, 2025-09-10:
also I didnt consider it when I first looked at your code, but its likely not the initial start of the SD card, but the sd scan task thats causing the crash, if you wanted to try anything deleting that function may fix it. This should be fixed in the new version (system runs in little chunks now) but to check you can try running the code without the scan task.