Jstudner/jcorp-nomad #67: Page content does not update due to caching

gwenvador, 2025-08-28:
When accessing the movies page after a new upload, changes to content are not reflected immediately. The browser seems to serve a cached version of the page and data. The only way to see updates reliably is to open the page in a new incognito window.
Is that on the user browser or could it be done on the server?

Jstudner, 2025-08-30:
The new experimental version (I'm so close to releasing this damn thing I've been doing stability fixes for a week) should fix this, it replaced the default caching system with a smart one that asks the backend if anything changed, also keeps track of watch progress in cache so you can resume per device.

Jstudner, 2025-09-08:
ok, I am still doing stupid stability fixes but I got this working fairly well. Need to add it to all of the pages, but I can have the frontend reference the index on load and remove items that are not there anymore / manualy grab new items without fully wipeing out the cache now. should work quite nicely.

Jstudner, 2025-09-16:
ok, this is sorta fixed. I am not going to close #67 yet, but the experimental branch has the basic of this functionality and I will be making improvments to it. main issue is it slows the page loading, I need to set it up to cache the info like before, but then after scanning update the page in a non-blocking way (spots differences and edits grid without effecting the existing entrys)

Robin-Sch, 2025-09-16:
Wow, thanks a lot man. Your work is highly appreciated

Jstudner, 2025-11-01:
Ok! Still havent gotten this into Main, but the experimental branch now does autocall requests on load, so its much snappier aswell, should provide for much quicker library updates that users wont notice (hopefuly).

closed 2025-11-01 by Jstudner