Jstudner/jcorp-nomad #81: text section exceeds available space in board

ethanocurtis, 2025-09-15:
when trying to flash the board i get the following:

Sketch uses 1423619 bytes (108%) of program storage space. Maximum is 1310720 bytes.
Global variables use 127052 bytes (38%) of dynamic memory, leaving 200628 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board

Jstudner, 2025-09-15:
You have to check the board options, your flash size is set to 4mb not 16mb so it thinks its exceeding the size. check the exact settings needed on the instructibles.

ethanocurtis, 2025-09-16:
this worked! however my browser is not redirecting me to the home screen and on the screen the sd card logo isnt ticked on like hte wifi logo, i did format to fat32 with the provded folder structure. its a 256gb sd card.

Jstudner, 2025-09-16:
what does the serial output say? Its likley not mounting the card for one reason or another

Jstudner, 2025-09-16:
Enjoy!! Hope you get good use out of it! 😁